adafruit/circuitpython-org #53: Have mpy-cross Available For Download

sommersoft, 2019-03-18:
We should have compiled versions of `mpy-cross` available to download. 

If I understand correctly, previous availability was done manually by adding it to a release's assets. Since they aren't built dynamically, I would guess that they could live here in `assets/`.

ATMakersBill, 2020-01-24:
This would be very helpful for our projects that use non-express boards

dhalbert, 2020-01-24:
GitHub Actions can run jobs in Linux, Windows, or MacOS. It says ARM too, so maybe we could build an RPi version (?).

sommersoft, 2020-01-24:
To clarify the environments available in GitHub Actions:

- [GitHub-hosted Virtual Environments](https://help.github.com/en/actions/automating-your-workflow-with-github-actions/virtual-environments-for-github-hosted-runners#supported-runners-and-hardware-resources)
```
Windows Server 2019
Ubuntu 18.04
Ubuntu 16.04
macOS Catalina 10.15
```

- [Supported operating systems for self-hosted runners](https://help.github.com/en/actions/automating-your-workflow-with-github-actions/about-self-hosted-runners#supported-operating-systems-for-self-hosted-runners)
```
Linux (Architectures: X64, ARM64, ARM)
Red Hat Enterprise Linux 7
CentOS 7
Oracle Linux 7
Fedora 29 or later
Debian 9 or later
Ubuntu 16.04 or later
Linux Mint 18 or later
openSUSE 15 or later
SUSE Enterprise Linux (SLES) 12 SP2 or later

Windows
Windows 7 64-bit
Windows 8.1 64-bit
Windows 10 64-bit
Windows Server 2012 R2 64-bit
Windows Server 2016 64-bit
Windows Server 2019 64-bit

MacOS
macOS 10.13 (High Sierra) or later
```

dhalbert, 2020-08-30:
mpy-cross binaries, built for every merged PR and release, are now in https://adafruit-circuit-python.s3.amazonaws.com/index.html?prefix=bin/mpy-cross/

dhalbert, 2020-08-30:
I'm thinking that links to the mpy-cross downloads should be on circuitpython.org somewhere, though, maybe on the libraries page??

dglaude, 2021-03-07:
Yes it should be on circuitpython.org or better referenced or this kind of thing take place: https://www.recantha.co.uk/blog/?p=20904

PR #741 (2021-08-19): update libraries page and header links
I did some rewriting on the libraries page:
- Added link to readthedocs for all libraries.
- Reworked the .mpy explanation and other text.
- Added link to Libraries page in Welcome guide in two places
- Mentioned CircUp.
- Mentioned and linked to the Community Bundle. This could be expanded later.
- Redid headings a bit

I also revamped the top links a bit, renaming "API to "Documentation" "Help" to "Forums", and labelling the Discord and RSS links more explicitly.
The mobile header was also updated.
OLD:
[image]
NEW:
[image]

Fixes #53 
Fixes #526 
Fixes [... cut]
closed 2021-08-19 by kattni