adafruit/circuitpython-org #98: /stats  for circuitpython.org

ptorrone, 2019-03-30:
would be good to use the github api for up-to-date stats on downloads and have a human-readable page here:

https://www.circuitpython.org/stats

OR

if it's easier, use the JSON file that runs download stats.
https://github.com/adafruit/circuitpython-org/blob/master/_data/files.json

extra extra credit, totally OK with putting google analytics there too so folks can see site traffic, etc.

closed 2019-04-05 by tannewt

ptorrone, 2019-04-05:
@jwcooper @tannewt this is great, would it be possible to display them by top downloads per group?

jwcooper, 2019-04-05:
@ptorrone what would the group be? Board? Release?

ptorrone, 2019-04-05:
the download column per group, that way each group is by top downloads?

[image]

ptorrone, 2019-04-05:
@jwcooper just talked with @tannewt each release would be sorted by downloads (most being at the top of each release).

tannewt, 2019-04-05:
My understanding is that the top release shown would be the one with the most total downloads (3.1.2 now I think.)

ptorrone, 2019-04-05:
i meant the latest release is still at the top, and within that, top downloads.

ptorrone, 2019-04-05:
OK! it appears that firefox as per @tannewt displays it correctly but it's chrome that does not, all this time... sorry @jwcooper for the comments here, so this is now a chrome display issue :(

here is what it looks like on chrome.

[image]

sommersoft, 2019-04-05:
> so this is now a chrome display issue

Also an Edge issue, as it displays "non-sorted" as well.

cpforbes, 2019-04-05:
The sort comparison function isn't returning the right thing. It should be:

`return b.downloads - a.downloads`
to get highest to lowest.

See https://developer.mozilla.org/en-US/docs/Web/JavaScript/Reference/Global_Objects/Array/sort#Description

jwcooper, 2019-04-05:
Pushed a pr for the fix: https://github.com/adafruit/circuitpython-org/pull/118

sommersoft, 2019-04-06:
Confirm that #118 fixes for Edge as well. Thanks @cpforbes & @jwcooper.