adafruit/circuitpython-org #106: disallow display list for /downloads by name

ptorrone, 2019-03-31:
would be a good feature to add a list of names in a file to not-display, something like...

disallow...
feather_radiofruit_zigbee
datalore_ip_m4
cp32-m4
gb_m4
pca10040

these can stay as we'll have some images soon...
sam32
meowmeow
catwan_usbstick

i'll do PRs for those soon.

tannewt, 2019-04-01:
Why? Those boards exist has much as the haxpresses do.

ptorrone, 2019-04-01:
@tannewt those are examples for disallow, i'd probably not display some of them until we had more info so there are no "unknown" slots without photo/image/text that do not link to anything, one specific example "feather_radiofruit_zigbee" unfinished board and will not have anything for a long time.

tannewt, 2019-04-02:
I don't think we need a control for this on the website side. Either we should get photos/info from folks or disable building the board from CircuitPython (which will prevent it from appearing on the website).

Datalore is @TG-techie, cp32 is @siddacious, gb_m4 is mine and the pca10040 is nRF supported by 3.x.

TG-Techie, 2019-04-02:
As someone who made one of those it sounds fine to me. If some wants their
board up on the site and has started distributing the board it would seem
reasonable to reach out to the admins on discord to enable it building.

ptorrone, 2019-04-02:
if it's a choice of having builds or not, i'd rather have the builds so folks can use/test their hardware. ideally the board url would be available like: https://circuitpython.org/board/feather_radiofruit_zigbee

but would not be displayed on: https://circuitpython.org/downloads

until we allow it when there are photos from the maker, etc.

does that work?

tannewt, 2019-04-03:
So we chatted about this and agreed we should add a boolean in the front matter that can cause a board to not appear on the /downloads page. The full link would still be available.

jwcooper, 2019-04-05:
Just to be clear, the boolean would be on the unknown.md file?

These boards are only listed in the _data/files.json from the build process and do not have their own board files to selectively toggle display on/off in the front matter.

We could also create simple placeholder board.md files each time a board is added to the build process, but it would likely need to be automated to keep ahead of it.

tannewt, 2019-04-05:
I think we should require a new .md file for the board. The flag would live in that file and basically designate it as a draft.

I'm ok with unknown.md being displayed as a way to see what needs to be supported. For boards we want as draft, we an add an .md before the release includes it.

closed 2019-04-08 by ptorrone