adafruit/circuitpython-org #113: Library Bundles

jedgarpark, 2019-04-03:
It would be sweet to have the library bundle download link next to the related .uf2 download button.

makermelissa, 2023-05-08:
Although convenient, the bundles are not board specific. Adding this to each board page may have the users thinking otherwise. Additionally, since the link changes each day, this may add additional overhead. Currently you can go to the Libraries link at the top of every page and download the bundles from there where having it in a central place makes updating the link easier. Is that acceptable?

tannewt, 2023-05-09:
I've just started using `circup` and would recommend it too.

makermelissa, 2025-12-18:
I'm going to close this as it isn't practical.

closed 2025-12-18 by makermelissa