adafruit/circuitpython-org #120: image missing from https://circuitpython.org/awesome

ptorrone, 2019-04-07:
image missing from https://circuitpython.org/awesome looks like ya'll are on it, here is the issue.

[image]

dhalbert, 2019-04-07:
This was fixed by #114, but somehow the site hasn't been rebuilt, which I don't understand.

sommersoft, 2019-04-08:
I can't speak to the site building after no.114 or not as I can't even see that side of things. But, could it maybe be related to the image source URL?

Its currently `https://github.com/adafruit/awesome-circuitpython/blob/master/awesome_circuitpython.png`. Navigating to that URL in a browser brings up the repo, not just the image.

Maybe the download URL would be better? `https://raw.githubusercontent.com/adafruit/awesome-circuitpython/master/awesome_circuitpython.png`

PR #123 (2019-04-08): update awesome
Fixes #120. This was supposed to have been fixed by #114, which has the right change, but didn't seem to change `master`. Let's try again.
closed 2019-04-08 by sommersoft