adafruit/circuitpython-org #153: /blinka page that works almost-like /downloads

ptorrone, 2019-04-15:
hi @jwcooper let me know if we need to chatter on vid, etc. here it goes.

we want to have /blinka be a listing of all the linux boards that support blinka. it would function the same way we do downloads.

another section on the repo /boards  is where we'd put the .md files to display, we'd use the front matter tag features: etc.. all the same, BUT we would not at this time need to have anything automatically coming in. we will manually add the board names, text, etc, etc.

@tannewt if i missed anything or how to explain this better, please comment in!

jwcooper, 2019-04-15:
If I understand correctly, we could do _blinka/board_name.md. Boards won't co-exist between /downloads and /blinka?

There won't be a files.json connection, right? Where do we link to on the download page for files, etc?

ptorrone, 2019-04-15:
> If I understand correctly, we could do _blinka/board_name.md. Boards won't co-exist between /downloads and /blinka?

correct!

> There won't be a files.json connection, right? Where do we link to on the download page for files, etc?

correct!

tannewt, 2019-04-15:
@jwcooper no files.json. We'll either want to link to Blinka: https://github.com/adafruit/Adafruit_Blinka or individual loading instructions because blinka may have unique deps per board.

jwcooper, 2019-04-19:
This was completed and merged. New files can be added to _blinka/

closed 2019-04-19 by jwcooper