adafruit/circuitpython-org #154: Include bootloader update downloads

deshipu, 2019-04-15:
Now that we have some bootloader updates, it would be nice to be able to point people to circuitpython.org to download them as well.

dhalbert, 2019-04-25:
This is getting more important due to the need to update the bootloaders because of the MacOS 10.14.4 MakeCode issue: That MacOS version cannot see the BOOT drive when there's a MakeCode program loaded

Note that the board names for the bootloader and CircuitPython are not the same, and there's not even a one-to-one correspondence: some bootloaders are used for multiple similar boards.

Racer1Tn, 2019-06-27:
There is a Windows problem that I cannot copy or read files from the Pygamer. I've tried different computer running Windows 10 and 8.1, different ports, and also various usb cables. I get the load file screen when connected via usb cable and the file explorer pops up. I get not responding after a time of trying to open or copy files to the pygamer. I've had similar issues with other Adafruit devices. However I did not have any issue doing the same with the Pyportal device. Soo frustrating.

dhalbert, 2019-06-27:
@Racer1Tn Please post in the support forums https://support.adafruit.com so we can help you there. This is for software bug issues.

dhalbert, 2020-04-15:
Fixed by #446.

closed 2020-04-15 by dhalbert