adafruit/circuitpython-org #155: Don't generate links to firmware builds that are missing

deshipu, 2019-04-15:
The circuitpython.org lists all boards that have descriptions, but not all of those boards are getting built automatically. That results in broken links to firmware downloads. We probably shouldn't link to firmware downloads that are not there.

jwcooper, 2019-10-04:
Sorry, I know this is quite old now, but do you have an example link you can provide?

tannewt, 2019-10-07:
I don't think this should happen. We have a check that all boards are included in the CI config and the releases have been reliable in getting all of the files posted.

theacodes, 2020-04-10:
Just confirmed that all board have working download links, closing this issue.

Note: some boards *do* link to an empty s3 folder for their "browse s3" button, but these seem to not have been built at all yet and don't have any other download links.

closed 2020-04-10 by theacodes