adafruit/circuitpython-org #159: Mirror of awesome-circuitpython is old

TheKitty, 2019-04-18:
The mirror of awesome-circuitpython is not current. The file was changed several days ago but the circuitpython.org copy did not change as it points to an older copy.

Change so that the current copy is always pulled
https://github.com/adafruit/awesome-circuitpython/blob/master/README.md

sommersoft, 2019-05-04:
@TheKitty,

What update interval would be acceptable for this?

If "realtime", then yeah, that would be somehow writing the page's code to retrieve at access. Which wouldn't update the mirrored submodule in this repo, and would render the submodule pointless. I am not in a position to espouse on feasibility of this though...webdev is definitely not my forté.

If something along the lines of weekly, then the submodule update could be worked into adabot. Adafruit_CircuitPython_Bundle updates work along these lines.

makermelissa, 2022-06-08:
Would a daily update be feasible to work into adabot or would that require approving a new PR each day?

jepler, 2022-06-09:
Fixed by #967

closed 2022-06-09 by jepler