adafruit/circuitpython-org #176: Download buttons missing from libraries page on desktop browsers

kattni, 2019-05-02:
@jwcooper The bundle download buttons are missing from the Libraries page, only on desktop browsers (mobile works). I verified it on Chrome on Linux, and Firefox on Mac and Linux.

If you could take a look, that would be great. Thanks!

[image]

jwcooper, 2019-05-02:
Do you have any ad-blocking software that is preventing the github api from loading (privacy badger, ertc)? Any errors in the javascript console? I tested all three major browsers, and it's loading OK (on mac OS).

tannewt, 2019-05-09:
I see the buttons on Mac with Firefox too.

kattni, 2019-05-14:
This has continued to happen inconsistently, but for the most part, it's not happening. I'm going to close this for now and if I keep running into it, I'll reopen.

closed 2019-05-14 by kattni