adafruit/circuitpython-org #299: Feature Request: Sort by Date Added

TheKitty, 2019-09-30:
Suggestion for CircuitPython.Org/downloads - sort by date board added - get the newest first

phillip torrone 11:59 AM
yah, add that as a feature request as an issue in github
would be, click the filter button
then..
Sort By
Downloads
Board Name (A To Z)
Board Name (Z To A)
adding Date (Most recent)

jwcooper, 2019-09-30:
This also sort of ties into the RSS feed issue (https://github.com/adafruit/circuitpython-org/issues/291) as we need a way to sort by newly added boards to add to the RSS feed. One option would be to add a date field to each _board/* file as we add them.

makermelissa, 2020-03-31:
First part of this is done: downloads via #437.
Next up is the Blinka boards.

PR #439 (2020-04-01): Added Date Sorting for Blinka boards
Added date Sorting for Blinka boards. Fixes #299. Went through and added "data_added" using initial GitHub PR dates. Cleaned up some console.log statements. This also fixes the PocketBeagle Name. Also fixes a Javascript error I found when loading Blinka page on live site:
```
Uncaught TypeError: Cannot read property 'length' of null
    at handlePageLoad (downloads.js:76)
    at HTMLDocument.<anonymous> (downloads.js:26)
```
I tested this locally and it seems to work very well.
closed 2020-04-01 by tannewt