adafruit/circuitpython-org #306: Add label data to libraries.json for inclusion in Contributing

kattni, 2019-10-09:
Please include the GitHub issue label data in the libraries.json file used to generate /libraries/contributing so we can include an option in the UI to filter by label.

sommersoft, 2019-10-10:
Pre-PR request for comment. @jwcooper, does this format work for you?
```
"Adafruit_CircuitPython_Bitmap_Font": [
      {
        "title": "TTF support (Open 147 days)",
        "url": "https://github.com/adafruit/Adafruit_CircuitPython_Bitmap_Font/issues/10",
        "labels": [
          "None"
        ]
      },
      {
        "title": "Remove backward-compatible imports when 4.0.0 released (Open 203 days)",
        "url": "https://github.com/adafruit/Adafruit_CircuitPython_Bitmap_Font/issues/9",
        "labels": [
          "Hacktoberfest",
          "good first issue"
        ]
      }
    ],
```

Old format for reference:
```
"Adafruit_CircuitPython_ADS1x15": [
      {
        "https://github.com/adafruit/Adafruit_CircuitPython_ADS1x15/issues/40": "Can cause crashes when used with multiple threads at once - thread-safety (Open 6 days)"
      },
```

jwcooper, 2019-10-11:
Yes, that looks good to me at first glance!

Would it be possible to name this new format something like libraries.v2.json so that we don't break the site while we transition to the new format?

sommersoft, 2019-10-11:
> Would it be possible to name this new format something like libraries.v2.json so that we don't break the site while we transition to the new format?

Definitely. Will make that happen when pulling in the updated adabot submodule.

PR #307 (2019-10-11): Add Issue Label Data to libraries.json
Fixes #306
closed 2019-10-11 by jwcooper