adafruit/circuitpython-org #317: add pictures and md page for sparkfun_qwiic_micro with and without flash

ladyada, 2019-10-24:
`@edspark I'll get the pictures and .md page added sometime tomorrow! Thanks @ladyada`

ladyada, 2019-10-24:
tagging @edspark here so they have the assignment

edspark, 2019-10-25:
Works for me, unfortunately this will have to wait till Monday, but it's at the top of my list. Have a great weekend!

edspark, 2019-11-04:
Pull request [here](https://github.com/adafruit/circuitpython-org/pull/321).

edspark, 2019-11-04:
Can we close this out since the pull request was accepted?

makermelissa, 2019-11-04:
Almost... I'm applying a fix and then we can...

makermelissa, 2019-11-04:
Ok, fixed.

closed 2019-11-04 by makermelissa

ladyada, 2019-11-04:
yay thanks folks - @edspark we needed to make two entries because there are two builds that were defined. please check https://circuitpython.org/downloads

makermelissa, 2019-11-04:
It also wanted very specific names for the boards.

edspark, 2019-11-04:
Thank you for updating that!