adafruit/circuitpython-org #330: https://www.circuitpython.org/ is insecure

tannewt, 2019-12-04:
Going to https://www.circuitpython.org/ generates a cert warning on Firefox.

ptorrone, 2019-12-04:
chrome too, here is some more info:
NET::ERR_CERT_COMMON_NAME_INVALID

[image]

jwcooper, 2019-12-04:
This is currently a known issue with GitHub pages. I'll see about setting up a cloudflare redirect or something similar.

Discussion here:
https://github.community/t5/GitHub-Pages/Does-GitHub-Pages-Support-HTTPS-for-www-and-subdomains/m-p/7202#M495
https://github.com/isaacs/github/issues/1675

tannewt, 2019-12-04:
@ptorrone keep this in mind when linking to the site. I saw the www. version recently in something (can't remember what.)

ptorrone, 2019-12-04:
@tannewt @jwcooper yep, looks like https://circuitpython.org/ is fine, so if anyone sees www somewhere and can edit, please do, otherwise let me know and i'll edit, or we'll see what we can do.

makermelissa, 2019-12-04:
I’m not familiar with the intricacies of Github pages, but on apache-based servers, you can add a rule that will forward somebody from www.circuitpython.org <http://www.circuitpython.org/> to circuitpython.org. Likewise with http to https. Is there a way of doing that with the GitHub pages?

Melissa

jwcooper, 2019-12-04:
Unfortunately, I haven't seen any access to anything like this yet. GitHub doesn't really offer a suggested solution. 

The best option I've seen is have www.circuitpython.org be setup with CloudFlare ssl and do a CloudFlare page rule to redirect it back to circuitpython.org. GitHub is using letsencrypt to create the ssl cert, and isn't creating one that's valid for www.circuitpython.org. They offer the option to do one or the other. 

There is a hack to swap the cname in github settings, but you'd have to remember to do that every 3 months as the certs expire, and I'd like to avoid doing that.

makermelissa, 2020-03-31:
@jwcooper I like your suggestion for the "best option". :)

theacodes, 2020-04-10:
Cloudflare is your best best here, I'm happy to help set it up if needed!

jwcooper, 2020-04-10:
We've started the process to transfer to cloudflare. Thanks for all the help everyone! We should be able to configure this once the transfer is complete.

jwcooper, 2020-04-10:
This has been completed. 

www.circuitpython.org will now 301 redirect to circuitpython.org.

closed 2020-04-10 by jwcooper

makermelissa, 2020-04-10:
Thank you @jwcooper