adafruit/circuitpython-org #385: Add learn guide links to all Adafruit boards

iayanpahwa, 2020-01-24:
It'll be convenient to have hyperlinked learn guides to Adafruit portal for contributed boards from Adafruit.

makermelissa, 2020-01-24:
Sounds like this would be similar to the Blinka pages. Good idea.

iayanpahwa, 2020-01-24:
@makermelissa As in these? [https://circuitpython.org/blinka](https://circuitpython.org/blinka). I can create a PR.

makermelissa, 2020-01-24:
Exactly.

iayanpahwa, 2020-01-25:
@rasponicc , I know you have been wanting to contribute to CircuitPython, would you like to take this one up as a good first issue? I’m certain maintainers will help you out when needed.

cc @makermelissa , @ladyada , @dhalbert

ladyada, 2020-01-25:
will definitely help, this is an excellent beginner issue :)

rasponicc, 2020-01-26:
Yes, will love to! Thanks!

iayanpahwa, 2020-01-26:
@makermelissa can you assign this issue to her? (@rasponicc)

rasponicc, 2020-01-28:
So learn guide exists on 1 board out of all(see: https://circuitpython.org/board/circuitplayground_express/). The other way of creating the learn guide is adding a section to the right side of the page(ref: file. look at the right hand side). Which way should I go about to add the learn guide across all board? I feel the first one, that is already there for **Circuit playground express** board on the website will work better in terms of quick readability, fast access(without the viewer having to look for it) on both pc and mobile platform.
[image]

iayanpahwa, 2020-01-28:
@rasponicc thanks for taking this up. I would like to see the approach in the screenshot having a dedicated button attracts the eye and visitor will not have to skim through a lot of texts where eyes often get lost. Also you may try both? Hyperlinking under Tutorials tab and making a dedicated session as can be seen in the screenshot, also I was thinking of hyperlinking the main board cover photo with the learn guide. We can have all three maybe!

The tutorials tab (on the left) may in future be used to add a lot of tutorials using that board from Adafruit's learning system and other users blog so having a dedicated button at the right looks a charm to eyes. 

thoughts? @ladyada , @makermelissa

rasponicc, 2020-03-10:
PR created #420

closed 2020-03-11 by tannewt