adafruit/circuitpython-org #399: Unknown Boards with CP5 Beta 5

makermelissa, 2020-02-05:
It looks like the following boards are missing:
ndgarage_ndbit6
meowbit_v121

Issue #396 is already open for the ndbit6, but since there hasn't been any movement on it, I may just try and get something up in the meantime.

makermelissa, 2020-02-05:
Ok, it appears `kittenbot_meowbit` was renamed to `meowbit_v121` since `kittenbot_meowbit` no longer appears in the assets. @hierophect, can you confirm?

hierophect, 2020-02-05:
@makermelissa I've always had it as `meowbit_v121` - where did `kittenbot_meowbit` come from? I'm happy to switch to that if needed.

makermelissa, 2020-02-05:
Thanks @hierophect. I probably saw it was named that in the circuitpython boards at some point. No problem, I can change it on my end.

PR #400 (2020-02-05): Updated a couple of unknown boards
Renamed meowbit to the correct name. Also, added a very basic template for ndbit6, though there is pretty much no searchable information I could find about it, but this is better than it saying unknown.

Fixes #399.
closed 2020-02-05 by tannewt