adafruit/circuitpython-org #410: Misleading photo for Arduino Nano 33 BLE

aidancuckoo, 2020-02-26:
The photo for the Arduino Nano 33 BLE is actually a photo of an Arduino Nano 33 BLE Sense, its sensor filled cousin. Would you mind replacing the photo?

ladyada, 2020-02-26:
hi, they are the same boards / pinouts. we tested specifically with the SENSE version.

closed 2020-02-26 by ladyada

dhalbert, 2020-02-26:
It might make sense (ouch) to update the description and/or board title in https://github.com/adafruit/circuitpython-org/blob/master/_board/arduino_nano_33_ble.md. Would you like to submit a PR?

tannewt, 2020-02-26:
Why don't we just make two board definitions?

ladyada, 2020-02-26:
thats fine too :)

aidancuckoo, 2020-02-26:
@tannewt that's what I was thinking! As for the LPS22, the description for the Fritzing part which was released says that it uses an LPS25 and not an LPS22. Maybe they're code compatible and @ladyada I noticed that they both use the same PCB too! So it's probably 90%/100% compatible with CircuitPython.

tannewt, 2020-02-26:
@aidancuckoo Would you like to add a PR to CircuitPython for it? There is a guide on how to do it here: https://learn.adafruit.com/how-to-add-a-new-board-to-circuitpython

aidancuckoo, 2020-02-27:
@tannewt One note - I won't be able to do any hardware testing as I don't have the boards. Maybe I'll say UNTESTED on the website?

aidancuckoo, 2020-02-27:
> @aidancuckoo Would you like to add a PR to CircuitPython for it? There is a guide on how to do it here: https://learn.adafruit.com/how-to-add-a-new-board-to-circuitpython
@tannewt Unfortunately I will only be able to put it on the website as I don't have the boards. Maybe I'll put "Untested" on the page?

tannewt, 2020-03-03:
@aidancuckoo I have a Nano Sense that I can test with and may have the other as well. No need to say untested.