adafruit/circuitpython-org #426: Include The Available Built-in Modules For Each Board

sommersoft, 2020-03-21:
In the CircuitPython documentation, we have a matrix that shows the available built-in modules for each board. https://circuitpython.readthedocs.io/en/latest/shared-bindings/support_matrix.html

I feel the current location and layout are cumbersome, involve lots of scrolling, and not easy to find. 

I would like to extend this information to circuitpython.org, so it's more visible to users. The matrix information in the documentation is built with a Python script, which is easily extended to use elsewhere.

I have some various implementation ideas:
- **Minimum**:
  - Available modules are simply listed on each board's page, and only apply to the current stable release.
  - Information resides in `_data/files.json`, and is delivered with the current updates to that file.
- **Intermediate**:
  - Includes **Minimum**, but includes available modules for both stable and current unstable releases.
- **The Full Monty** 🐍:
  - Includes **Minimum** and **Intermediate**, but in a way that also allows for filtering on the downloads page based on available module(s). However, this could be tricky and/or confusing with having both stable and unstable version information.
  - Might need the information to live outside of `_data/files.json`, though I expect not with the current Jinja templating.

As a frontend-design-challenged individual, I defer all of that to the experts. 😄

FoamyGuy, 2020-03-23:
I would definitely like to be able to see this info on the board pages. I think the minimum or intermediate proposals would be a great start. Filtering sounds cool but I'm not certain if I'd use it personally. I tend to go looking if a certain device supports a certain thing more than which devices do support this certain thing. But who knows, if it were implemented maybe I'd use it more than I think.

makermelissa, 2020-04-01:
Ok, after dabbling more with the site, I think the intermediate option would probably be the best one. Another option not mentioned is a hybrid of the minimum and full monty which gets around the challenge of the filtering options. If we went with the Full Monty as you proposed, we could combine the list of modules if we wanted to have the filtering and then the user could choose a build that has the module they want. They may just be out of luck if they are looking for multiple features and one is on the stable release and the other is on the unstable. I still lean towards intermediate as a starting point because even if you include all the information, we can still start by showing just the stable.

sommersoft, 2020-06-27:
Finally jumped into this. This is a draft I settled on, with manually putting in some info into `_data/file.json`. Wording and layout could probably be better, but again...not my forte. I chose the coloring to further highlight what is available in stable compared to unstable (if differences occur).
[image]

Currently working on hooking `docs/shared_bindings_matrix_.py` into `tools/build_board_info.py` in CircuitPython repo, so the info is updated with releases.

PR #493 (2020-06-30): Include The Available Built-in Modules For Each Board
Fixes #426 

Example of page render here: https://github.com/adafruit/circuitpython-org/issues/426#issuecomment-650640474

@jwcooper, please let me know if my CSS edits aren't correct. I've never used Sass before, so I took a guess...

Updates to `_data/files.json` were "manual" for this PR, but the release PRs from the core repo should include future changes. See [circuitpython #3087](https://github.com/adafruit/circuitpython/pull/3087).
closed 2020-06-30 by tannewt