adafruit/circuitpython-org #431: Add UARTLogger2 to circuitpython.org website

jepler, 2020-03-26:
Added to circuitpython 5.1.0 https://github.com/adafruit/circuitpython/pull/2705

@nightskysk can you help us out?  There's a guide here: https://learn.adafruit.com/how-to-add-a-new-board-to-the-circuitpython-org-website

makermelissa, 2020-03-26:
@NightSkySK, if you could at least provide a good photo and some information on the board, I could put something together, but if you're able to submit a PR, that'd even be better. Thanks.

makermelissa, 2020-03-31:
Since there's no info that I can add, I'll create a hidden stub for this so it doesn't show up on the website as unknown for now.

makermelissa, 2020-04-21:
This is now hidden. @NightSkySK please submit a PR if you want it shown to the world. Closing for now.

closed 2020-04-21 by makermelissa