adafruit/circuitpython-org #432: Add TG-Watch02A to circuitpython.org webiste

jepler, 2020-03-26:
Added to circuitpython 5.1.0 https://github.com/adafruit/circuitpython/pull/2695

@TG-techie can you help us out?  There's a guide here: https://learn.adafruit.com/how-to-add-a-new-board-to-the-circuitpython-org-website

makermelissa, 2020-03-26:
@TG-Techie, if you could at least provide a good photo and some information on the board, I could put something together. Thanks.

TG-Techie, 2020-03-28:
@makermelissa  I'll gladly try to add it. should i discord you if i have any trouble?

makermelissa, 2020-03-28:
Thanks @TG-Techie. Just reply to this issue if you do.

makermelissa, 2020-03-31:
I'll create a hidden stub for this so it doesn't show up on the website as unknown for now.

makermelissa, 2020-04-21:
This is now hidden. @TG-Techie please submit a PR if you want it shown to the world. Closing for now.

closed 2020-04-21 by makermelissa