adafruit/circuitpython-org #497: Missing Version 6.x library bundle

dglaude, 2020-07-05:
> You can always find the latest releases of the libraries bundle on GitHub.
> Bundle Version 5.x
> This bundle is built for use with CircuitPython 5.x.x. If you are using CircuitPython 5, please download this bundle.

- Is it OK to use Version 5.x for CP 6.x?
- Is Version 5.x any way different from Version 4.x?
- Does that only change with MPY format change or there are other possible area if incompatibility such as depreciated and now removed things in I2C API?
- Also should we still show Version 4.x?

dhalbert, 2020-07-05:
> * Is it OK to use Version 5.x for CP 6.x?

Yes.

> * Is Version 5.x any way different from Version 4.x?

No.
> * Does that only change with MPY format change or there are other possible area if incompatibility such as depreciated and now removed things in I2C API?

Right, it would be for a .mpy change, but there hasn't been one for a while. We have talked about having different bundles for different versions, but haven't come up with a mechanism yet. Instead, right now we add compatibility code in the libraries that need it.

https://github.com/adafruit/Adafruit_CircuitPython_Bundle/issues/181
https://github.com/adafruit/Adafruit_CircuitPython_Bundle/issues/175

> * Also should we still show Version 4.x?

We will probably drop 4.x soon. that we are doing 6.x test releases.

tannewt, 2020-07-13:
6.x versions of the bundle are now available: https://github.com/adafruit/Adafruit_CircuitPython_Bundle/releases/tag/20200711

closed 2020-07-13 by tannewt