adafruit/circuitpython-org #501: espruino_wifi.md Wifi might not work with CP.

dglaude, 2020-07-13:
I found this board: https://circuitpython.org/board/espruino_wifi/

However, I am pretty sure there are strong limitation in using the Wifi with CP on this board.
The reason is that it is 8266 based and not ESP32 SPI Airlift based.

Maybe this should be indicated more clearly on the download page.

More generally, there are board that are only partially supported, but if someone use the download page to pick his next board, it could lead to acquiring a board with poor CP support.

I believe there should be a flag or indication to distinguish the board that are fully supported, the board with early support but still missing some things in the software (like STM32 and the rotaryio), and the board that have advertised functionnality but that do not really work and might never work fully.

tannewt, 2020-07-13:
Please make a PR to add a note to the board page: https://github.com/adafruit/circuitpython-org/blob/master/_board/espruino_wifi.md

This file has a similar warning for USB: https://github.com/adafruit/circuitpython-org/blob/master/_board/espressif_saola_1_wroom.md

It should probably warning WiFi is unsupported as well.

dglaude, 2020-08-03:
Apparently I fixed this on my fork, but forgot to do a PR. The PR is here now: #514

makermelissa, 2020-08-08:
This was merged, so closing.

closed 2020-08-08 by makermelissa