adafruit/circuitpython-org #519: Adding spanish translation to the site

fede2cr, 2020-08-10:
I think I would be of great use to the spanish community, if the circuitpython site was translated to spanish.

I'm happy to do the translating of the content, and if need with researching plugins for the website, etc.

tannewt, 2020-08-10:
@fede2cr Please do! I'm not sure the process for translation with Jekyll so please do whatever works for you. Thanks!

kattni, 2020-08-10:
@jwcooper Here is the translation offer you and I talked about the other day.

@fede2cr It'll be best for you to work with Justin on how to do this.

jwcooper, 2020-08-11:
I've been looking into plugins that might help us out. Tricky part is maintaining all of our existing links.

Will test this plugin out first: https://github.com/kurtsson/jekyll-multiple-languages-plugin

makermelissa, 2020-08-11:
I do have some experience with doing this on a PHP-based website. I'm not sure how well that would transfer to Jekyll, but we maintained phrase lists for each language, similar to how we do it with CircuitPython, and just substituted it in the phrases. For missing phrases, we could use English as a fallback such as adding new boards.

So there's really 2 parts to this. The first is the actual translations for the page, which I think would be the easier part. I think it would be similar to how we have the same download page for each board, but swap in words. The second part is the UI for switching between languages, if you want that capability.