adafruit/circuitpython-org #524: Downloads Page - Unable to link to a feature with a dash

jwcooper, 2020-08-20:
https://circuitpython.org/downloads?features=Feather-Compatible

That link should work, but ends up back at /downloads after page load (the feature is filtered though).

If you also go to /downloads and select the "Feather-Compatible" feature, it will not populate the url with the selected feature.

jwcooper, 2020-08-20:
Another issue is a url with special characters, all sorts of issues there. Unable to deselect the feature once set. Page won't load and filter the feature, etc.

Example:
https://circuitpython.org/downloads?manufacturers=Sean+%27xobs%27+Cross

makermelissa, 2020-10-02:
Ok, it looks like I _may_ have accidentally fixed the features link issue with #552. It's still not populating the URL and the special characters still appears to be an issue.

closed 2021-12-01 by makermelissa