adafruit/circuitpython-org #575: Add the MagTag board

dglaude, 2020-11-09:
The MagTag is not visible yet on CircuitPython.org

Waiting for this pull request to be merged: https://github.com/adafruit/circuitpython/pull/3653
I think that right now, the firmware should be here: https://adafruit-circuit-python.s3.amazonaws.com/index.html?prefix=bin/adafruit_esp32s2_eink_portal/

But hope things to be easier by the time my MagTag reach me. :-)

ladyada, 2020-11-09:
yes you can use that firmware for now, the name will be updated

dglaude, 2020-11-10:
#579 seems to solve this for me: https://circuitpython.org/board/adafruit_magtag_2.9_grayscale/
Now we just need a 6.0.0 release with MagTag in so that we can have one firmware ready to download without going to S3: https://adafruit-circuit-python.s3.amazonaws.com/index.html?prefix=bin/adafruit_magtag_2.9_grayscale/

closed 2020-11-10 by dglaude