adafruit/circuitpython-org #597: Default language selection for firmware download?

dglaude, 2020-11-20:
There is something strange on /download from circuitpython.org. When I want to download CircuitPython 6.1.0-beta.1 it default to French (my browser is maybe configured for advertising I prefer french content) but 6.0.0. is proposed in English.
I have the "problem" in Firefox and Chrome... but not on Edge.
As if the page was smart, but only for one of the version.

Not a big deal, maybe an annoyance when by mistake I download a French firmware.

dglaude, 2020-11-20:
It is somewhere in the javascript:
- When I block javascript for the page it stay in English.
- When I activate javascript, the page first show English and then switch to French.

dglaude, 2020-11-20:
This activate when my browser connect to accounts.adafruit.com and if I browse in an anonymous browser, I don't have that behaviour.
This also explain that I don't have the behaviour in Edge, I never connected to Adafruit.com with Edge (Chromium version).
It might also be linked to 2Factor, because I enabled that recently, and do not remember that behaviour.

Maybe why is it a bug you may ask?
The page should change both language selection (for 6.0.0 and 6.1.0 beta) or none.

There is a download.js that contain code that check language things on load:
```
document.addEventListener('DOMContentLoaded',function() {
  var languageSelect = document.querySelectorAll(".language-select select");
  languageSelect.forEach(function(select) {
    select.onchange = languageSelectHandler;
  });

  var script = document.createElement('script');
  script.setAttribute('src', '//accounts.adafruit.com/users/locale?callback=setLocale');
  document.body.appendChild(script);
},false);

```

tannewt, 2020-11-24:
Weird! The code to generate the HTML is identical: https://github.com/adafruit/circuitpython-org/blob/master/_includes/download/board.html#L7

dglaude, 2020-11-24:
I have not looked everywhere as Javascript is not my thing.
What I did not figured out either is that nowhere on my adafruit account I say "French".
So I am curious to know what is going on. :-)

jwcooper, 2020-11-24:
@dglaude, what is returned from this request in your scenarios from above:

https://accounts.adafruit.com/users/locale.json

We're simply using the Accept-Language header sent by the browser. We have to use that accounts.adafruit.com callback, as this is a statically built site and we're not able to get the headers easily otherwise. It's not loading anything from an adafruit account, which should simplify debugging this issue for us.

I'm looking into if there is a javascript issue beyond that.

dglaude, 2020-11-24:
At troubleshooting time, I changed my browser Accept-Language to German... but the consequence of that on my internet experience were so hard on me (I don't understand German) that I gave up on that silly idea.

If booth language selection box use the same code, then it must be the javascript that only modify one and not the other. Because there are two language drop box, one for Stable and one for next release.

dglaude, 2020-11-24:
Installing Adafruit library is as simple as getting the zip file with all the MPY from https://circuitpython.org/libraries
Copy what you need in your lib folder.
When you have no more import error, this mean you have the library you need.
You can also get matching example (simpletest) from the other example-bundle.

Those setup.py and instruction in all the GitHub repo are for installation with Blinka, that is with the adaptation layer to use CircuitPython library on Raspberry Pi or other SBC. At least that is what I understand because I mostly ignore them, except if there is demo code.

Neradoc, 2021-01-10:
My languages are ["en-US","en","fr-FR","fr"].
And the menu for the stable is always "english" while the menu for the unstable is always "french" by default.

Looking into the selector javascript in `assets/javascript/download.js` I found that it takes the first language that matches any of the languages of the user.
```javascript
if (languages.includes(options[i].dataset.locale)) {
  // do stuff
  break
}
```
That's not good, languages are supposed to be in order of preference, so it should not find the first port that matches any of the users preferences, but prioritize the user languages in order. Basically, iterate over the user languages, and then only find the first match in the port's available languages. I believe this should fix the issue.

But that in itself would not explain the discrepancy in the two menus. As it turns out the menus are not identical, the order in which the languages are displayed is different.
Upon closer inspection, languages in the _data/files.json file are in a different order depending on whether we are looking at the stable or unstable release. I have no idea how those are generated, and if the order is deliberate (doesn't look like it), but they should probably be sorted in alphabetical order before looping over them. (Again, a different issue though)

dglaude, 2021-01-10:
I keep on downloading by mistake French firmware.
I much prefer one URL one content and have to explicitly choose the language, not a dynamic default.
When I have to upgrade 10 or 20 different boards, and for each of those I need to double check before downloading.

Neradoc, 2021-01-10:
Yeah a menu for global language selection that overrides the auto selection would be great.
It could work by using javascript's localStorage to store that preference from page to page.
Ideally I would prefer it to be set through the URL, but with a static site it would be a little complicated.

Or we could be clever and silently remember the last downloaded language and set it as default (via localStorage). It could be a little confusing but would probably work well in most cases if not all. Somebody would select a download language once and later maybe notice "oh it remembered, cool !". (Or "oh it remembered, creepy !", but...) If you download a different language only once you would only need to switch back once. Only if you flip flop between languages every time would it never be the correct one by default. Use two browsers !

dhalbert, 2021-01-10:
Could it just be a cookie?

makermelissa, 2021-01-25:
@dglaude could you check this again? I'm hoping #616 fixed the issue.

dglaude, 2021-01-25:
So current behavior is that both checkbox are at the same language, that language is French for me, when you change one, both change. If you close the browser and return, they keep the value you previously selected.

So this is as good as it can be, I don't have to worry anymore of downloading the wrong thing.

closed 2021-01-25 by dglaude