adafruit/circuitpython-org #627: Need to update bundles with adabot

makermelissa, 2021-01-25:
It appears the requirement to update Nokogiri to the latest has caused the bundles to need updating. According to the GitHub files it looks like something needs to be be run through adabot which I don't have access to. So would somebody with access to adabot please run it?

TheKitty, 2021-01-25:
Please escalate this request - I am in need of the site being updated for today's publishing of the Python on Microcontrollers Newsletter.

makermelissa, 2021-01-25:
After speaking with @kattni, I realized I was confused by the naming (and the fact I just woke up possibly). Also, here's a link to the actual failure: https://github.com/adafruit/circuitpython-org/runs/1762887363.

closed 2021-01-25 by ladyada

TheKitty, 2021-01-25:
So is this a non-issue or it still needs repair - the boards on the website are still not displaying

PR #628 (2021-01-25): Fix dependency issue
Fixes #627.