adafruit/circuitpython-org #638: Getting started guides for ESP32 boards need "how to flash the binary" info

mikew67, 2021-02-16:
Apparently the documentation team hasn't caught up with the "Getting Started" Guides on ESP32-related boards.  In contrast with other non-ESP32 boards, the information on how to load the CircuitPython builds (UF2 or bin) doesn't seem readily accessible.  Perhaps the following will help.  It's an annotated log of my process for loading a new CP release on an ESP-S2 Saoli board.  It's possible to use it raw, but one of your tech writers will probably want to use it as background for more polished instruction.  Or perhaps you have a better alternative.
Hope this helps.
Mike

========================================
mike@Mikes-MacBook-Pro-10-1 ESP32-S2 Saola binaries % date
Tue Feb 16 13:17:31 PST 2021

mike@Mikes-MacBook-Pro-10-1 ESP32-S2 Saola binaries % ls -al /dev/cu.S*
crw-rw-rw-  1 root  wheel   22,   5 Feb 16 13:13 /dev/cu.SLAB_USBtoUART

mike@Mikes-MacBook-Pro-10-1 ESP32-S2 Saola binaries % esptool.py --port /dev/cu.SLAB_USBtoUART chip_id
esptool.py v3.0
Serial port /dev/cu.SLAB_USBtoUART
Connecting....
Detecting chip type... ESP32-S2
Chip is ESP32-S2
Features: WiFi, ADC and temperature sensor calibration in BLK2 of efuse
Crystal is 40MHz
MAC: 7c:df:a1:0a:65:be
Uploading stub...
Running stub...
Stub running...
Warning: ESP32-S2 has no Chip ID. Reading MAC instead.
MAC: 7c:df:a1:0a:65:be
Hard resetting via RTS pin...

mike@Mikes-MacBook-Pro-10-1 ESP32-S2 Saola binaries % esptool.py --port /dev/cu.SLAB_USBtoUART write_flash 0 adafruit-circuitpython-espressif_saola_1_wrover-en_US-6.2.0-beta.2.bin
esptool.py v3.0
Serial port /dev/cu.SLAB_USBtoUART
Connecting....
Detecting chip type... ESP32-S2
Chip is ESP32-S2
Features: WiFi, ADC and temperature sensor calibration in BLK2 of efuse
Crystal is 40MHz
MAC: 7c:df:a1:0a:65:be
Uploading stub...
Running stub...
Stub running...
Configuring flash size...
Compressed 1359872 bytes to 861008...
Wrote 1359872 bytes (861008 compressed) at 0x00000000 in 75.9 seconds (effective 143.3 kbit/s)...
Hash of data verified.

Leaving...
Hard resetting via RTS pin...

mike@Mikes-MacBook-Pro-10-1 ESP32-S2 Saola binaries % echo "Move Saoli 1 to USB connection via wiring harness"
Move Saoli 1 to USB connection via wiring harness
mike@Mikes-MacBook-Pro-10-1 ESP32-S2 Saola binaries % echo "It will show up as CIRCUITPY thumbdrive"
It will show up as CIRCUITPY thumbdrive

mike@Mikes-MacBook-Pro-10-1 ESP32-S2 Saola binaries % cd /Volumes/CIRCUITPY
mike@Mikes-MacBook-Pro-10-1 CIRCUITPY % ls -al
total 52
drwxrwxrwx@ 1 mike  staff  16384 Feb 16 12:59 .
drwxr-xr-x  9 root  wheel    288 Feb 16 13:23 ..
-rwxrwxrwx  1 mike  staff      0 Dec  4  2016 .Trashes
-rwxrwxrwx  1 mike  staff   4096 Feb 16 13:12 ._boot_out.txt
-rwxrwxrwx  1 mike  staff   4096 Jan 31 14:47 ._code.py
drwxrwxrwx  1 mike  staff    512 Dec  4  2016 .fseventsd
-rwxrwxrwx  1 mike  staff      0 Dec  4  2016 .metadata_never_index
-rwxrwxrwx@ 1 mike  staff     82 Dec  4  2016 boot_out.txt
-rwxrwxrwx@ 1 mike  staff     22 Dec  4  2016 code.py
drwxrwxrwx  1 mike  staff    512 Dec  4  2016 lib

mike@Mikes-MacBook-Pro-10-1 CIRCUITPY % cat boot_out.txt
Adafruit CircuitPython 6.2.0-beta.2 on 2021-02-11; Saola 1 w/Wrover with ESP32S2

mike@Mikes-MacBook-Pro-10-1 CIRCUITPY % echo "Connect serial terminal for REPL access to USB serial port at 115200 baud"
Connect serial terminal for REPL access to USB serial port at 115200 baud
mike@Mikes-MacBook-Pro-10-1 CIRCUITPY %

=================================================================
Log of serial terminal output
=================================================================

Press any key to enter the REPL. Use CTRL-D to reload.

Adafruit CircuitPython 6.2.0-beta.2 on 2021-02-11; Saola 1 w/Wrover with ESP32S2
>>> print("Hello, World! From CircuitPython 6.2 beta")
Hello, World! From CircuitPython 6.2 beta
>>>
soft reboot

Auto-reload is on. Simply save files over USB to run them or enter REPL to disable.

Press any key to enter the REPL. Use CTRL-D to reload.

Adafruit CircuitPython 6.2.0-beta.2 on 2021-02-11; Saola 1 w/Wrover with ESP32S2
>>>

evildave666, 2021-11-02:
Boilerplate has been added to download pages on circuitpython-org by #753 on how to flash the bootloader on esp32-s2 devices, so perhaps this issue is redundant at this point.

makermelissa, 2022-05-26:
Closing because instructions have been added.

closed 2022-05-26 by makermelissa