adafruit/circuitpython-org #684: Include frozen CircuitPython libraries on board-specific Downloads page

kattni, 2021-04-27:
With the Project Bundle including everything in the zip file, we need a more visible place where we can point folks to show them what libraries are frozen into certain CircuitPython builds. This information is currently in the [mpconfigboard.mk file](https://github.com/adafruit/circuitpython/blob/main/ports/atmel-samd/boards/adafruit_neokey_trinkey_m0/mpconfigboard.mk) in a given CircuitPython board definition. Currently, built-in _modules_ are shown on the circuitpython.org/downloads pages for boards, but frozen libraries are not. The frozen libraries should be included as a separate list.

[This CircuitPython issue](https://github.com/adafruit/circuitpython/issues/4672) must be completed first. That will generate the list for displaying on the applicable Downloads pages. Once we have the data available, we can sort out displaying it on circuitpython.org/downloads/board_name/.

makermelissa, 2022-03-31:
> I think it may be a broader issue than this. Frozen libraries may change over time, and this may be a point of confusion for new and existing users (we had confusion over this with a PyPortal issue on Discord).

Are you referring to the fact that we may have CircuitPython 7.2.3 on a board with one set of frozen libraries and 8.0.0-alpha1 with a different set?

anecdata, 2022-03-31:
More in general, that freezing the portal and networking libraries is a recent addition, and we may freeze (or unfreeze) more (or different) as time goes on. It's probably not on many new and existing users' radars to pay attention to this, or know how to run a new feature that's not frozen yet.

Neradoc, 2022-04-17:
Note that, with builtin modules being removed and sometimes added back to some builds to make space for stuff, it's already changing with the version of Circuitpython, I think it's okay if the page only reflects the current release. I do second creating some reference on frozen libraries.

The frozen libraries could be in a separate list below the builtin modules on each board's page, under their actual module names. The generated list would need an update to get the names though. ([The matrix code in PR6288 does that](https://github.com/adafruit/circuitpython/pull/6288)).
For example:
- `adafruit_circuitplayground` instead of `Adafruit_Circuitpython_CircuitPlayground`
- `stage` and `ugame` instead of `pygamer` (from: `FROZEN_MPY_DIRS += $(TOP)/frozen/circuitpython-stage/pygamer` which contains stage.py and ugame.py)

makermelissa, 2022-04-18:
> More in general, that freezing the portal and networking libraries is a recent addition, and we may freeze (or unfreeze) more (or different) as time goes on. It's probably not on many new and existing users' radars to pay attention to this, or know how to run a new feature that's not frozen yet.

I think the idea was that just like with the included modules, it would be an automatic part of the CircuitPython release, so as libraries are frozen or unfrozen, it would be reflected here. While true that some users won't pay attention to that, it's useful for the folks that do.

I also release that it would likely be grouped per release like with modules, so having 2 different versions shouldn't matter.

Neradoc, 2022-04-19:
Oh I just realized that by adding the frozen modules to the matrix, they will automatically appear in the list of modules here, not separated from the builtin modules. Which, if it's ok, is fine and that implements that issue. But if we want to separate them on circuitpython.org (I think I'd prefer that), some additional work will be needed.

makermelissa, 2022-04-19:
Good point since they're added/removed the same way. I'm thinking if it has _CircuitPython_ (with the addition of neopixel and simpleio), it constitutes a library, which seems like the simplest way of doing it.

makermelissa, 2022-05-26:
This is done. See https://circuitpython.org/board/adafruit_magtag_2.9_grayscale/ for example.

closed 2022-05-26 by makermelissa