adafruit/circuitpython-org #685: Add search boards via primary processor.

water5, 2021-05-04:
From circuitpython.org/downloads page, add primary processor used on board option to 'Filters' likes 'SAMD21, SAMD51, ESP32-S2, STM32, nRF52840, RP2040 ...'.
Currently, can't to do that, e.g. : input 'RP2040' to 'Search for CircuitPython boards' field, the 'Pico' can't list out, 
input 'ESP32-S2' to 'Search for CircuitPython boards' field, 'Saola 1 w/WROVER, Saola 1 w/WROOM' can't list out.

makermelissa, 2021-05-05:
Huh, this is an interesting idea. I wonder if we can add something to the data files that CircuitPython generates since every board is already split up in Ports folders.

makermelissa, 2021-05-05:
Although SAMD21 and SAMD51 are grouped together in CP, so maybe not the best approach. If we add to the data files, I think we'll have to have some checks in place so the values are restricted to a list because Jekyll likes to group things like "SAMD21", "samd21", and "samd 21" differently.

tannewt, 2021-05-06:
In the long term I'd like to see a TOML database for boards similar to https://github.com/adafruit/nvm.toml. That info is more broadly useful than just having it in CP.

dhalbert, 2024-07-23:
This idea came up again to me while doing #1439. The "Processor family" values right now are inconsistent in level of detail: "esp32c6" exists, but not "samd51", only "atmel-samd". Searching for "samd51" does not bring up Metro M4, for instance, only boards with "SAMD51" in their board name.

makermelissa, 2024-07-23:
> This idea came up again to me while doing #1439. The "Processor family" values right now are inconsistent in level of detail: "esp32c6" exists, but not "samd51", only "atmel-samd". Searching for "samd51" does not bring up Metro M4, for instance, only boards with "SAMD51" in their board name.

Hmm, yeah. I think the reason we have the ESP chips more granular is because for chips like ESP32-S2 and ESP32-S3, we have a bootloader available, but for chips like the ESP32-C3, we don't.

tannewt, 2025-12-16:
I bet claude code would do a decent job if we pointed it to the circuitpython source too.

makermelissa, 2025-12-17:
What did you have in mind for how Claude code could help with this?

tannewt, 2025-12-17:
I bet claude could add the additional chip metadata to the cp-org files when pointed to the CP board source.