adafruit/circuitpython-org #711: Move from master to main default branch.

kattni, 2021-06-07:
We need to move the circuitpython-org repo to using the `main` default branch. There is clearly a lot involved here - so I'm opening this issue for discussion so we can be properly prepared.

Justin, I think you're the most familiar with this repo and the situation surrounding it. Dylan and I can provide what we learned from moving the CircuitPython libraries.

jwcooper, 2021-06-07:
We'll need to update at least the following:
https://github.com/adafruit/circuitpython-org/blob/2b36bba74b6820efbefa933ee7bd9a0a0abadf9e/.github/workflows/build.yml#L6

Every board file references master, but will redirect, at least for a while:
https://github.com/adafruit/circuitpython-org/blob/aa754d94214a68d314117ca9f975af2a9eca9a11/_board/aloriumtech_evo_m51.md#contribute

@kattni We should also pull anyone in that is familiar with how adabot commits to the repo, as that may need to get updated as well.

kattni, 2021-06-07:
@jwcooper I'll see what I can do regarding understanding adabot.

kattni, 2021-06-07:
@jwcooper Ok, regarding adabot, it should be good to go, as long as we keep the submodule updated (presumably especially when we move adabot to `main`). 

CircuitPython's release generators (https://github.com/adafruit/circuitpython/blob/c4f685aab1146237917fa36053d7626485330ca9/tools/build_board_info.py#L146 and https://github.com/adafruit/circuitpython/blob/c4f685aab1146237917fa36053d7626485330ca9/tools/build_board_info.py#L221) will need to be updated following the circuitpython-org move. 

`circuitpython-build-tools` appears good to go as well.

dhalbert, 2021-12-13:
Fixed by #712.

closed 2021-12-13 by dhalbert