adafruit/circuitpython-org #752: New "Install UF2 Bootloader" section in download/board.html?

durapensa, 2021-09-08:
[_includes/download/board.html](/adafruit/circuitpython-org/blob/master/_includes/download/board.html) contains a section `<div class="section unrecommended"><h3>UF2 Bootloader</h3>...` with conditionals  `{% if bootloader_board.family == 'nrf52840' %}` and `{% if bootloader_board.family == 'atmel-samd' %}` that provides *bootloader updating* instructions.

For any board with with TinyUF2 in flash, a section to *install bootloader* would be helpful, particularly for those boards that do not have TinyUF2 factory-flashed. "Install UF2 Bootloader" sections exist in Adafruit Learning System articles for [FunHouse](https://learn.adafruit.com/adafruit-funhouse/install-uf2-bootloader), [MagTag](https://learn.adafruit.com/adafruit-magtag/install-uf2-bootloader) and [Metro ESP32-S2](https://learn.adafruit.com/adafruit-metro-esp32-s2/install-uf2-bootloader), for the case of bootloader corruption or overwrite (e.g. if the board is esptool-flashed with Arduino firmware and the user wishes to return to using CircuitPython).

For the [Lolin S2 Mini](https://circuitpython.org/board/lolin_s2_mini/), I wrote a "Flashing UF2 Bootloader" section in  [_board/lolin_s2_mini.md](/adafruit/circuitpython-org/blob/master/_board/lolin_s2_mini.md), but would a new section in [_includes/download/board.html](/adafruit/circuitpython-org/blob/master/_includes/download/board.html) better handle cases like this one?

dhalbert, 2021-09-08:
If it can be described in a general way for all ESP32-S2 boards, then yes, that would be great, thanks! We would certainly entertain a PR.

durapensa, 2021-09-08:
I made a start on this and found that many boards are named differently in the respective `ports/esp32s2/boards` directories:

```
TINYUF2                           CIRCUITPYTHON        
adafruit_feather_esp32s2          adafruit_feather_esp32s2_nopsram 
adafruit_feather_esp32s2_tft      adafruit_feather_esp32s2_tftback_nopsram                         
adafruit_funhouse_esp32s2         adafruit_funhouse                
adafruit_magtag_29gray            adafruit_magtag_2.9_grayscale            
                                  ai_thinker_esp_12k_nodemcu                  
                                  crumpspace_crumps2               
                                  electroniccats_bastwifi          
espressif_hmi_1                   espressif_hmi_devkit_1                        
                                  espressif_kaluga_1.3             
                                  franzininho_wifi_wroom           
                                  franzininho_wifi_wrover          
gravitech_cucumberRIS_v1.1        gravitech_cucumber_m             
                                  gravitech_cucumber_ms            
                                  gravitech_cucumber_r             
                                  gravitech_cucumber_rs            
olimex_esp32s2_devkit_lipo_vB1    
                                  odt_pixelwing_esp32_s2           
muselab_nanoesp32-s2_wroom        muselab_nanoesp32_s2_wroom       
muselab_nanoesp32-s2_wrover       muselab_nanoesp32_s2_wrover      
targett_mcb_wroom                 targett_module_clip_wroom        
targett_mcb_wrover                targett_module_clip_wrover       
```

But these boards could work on an initial commit:

```
adafruit_metro_esp32s2            adafruit_metro_esp32s2   
artisense_rd00                    artisense_rd00                   
atmegazero_esp32s2                atmegazero_esp32s2    
espressif_kaluga_1                espressif_kaluga_1  
espressif_saola_1_wroom           espressif_saola_1_wroom          
espressif_saola_1_wrover          espressif_saola_1_wrover         
lilygo_ttgo_t8_s2_st7789          lilygo_ttgo_t8_s2_st7789         
lolin_s2_mini                     lolin_s2_mini                    
microdev_micro_s2                 microdev_micro_s2                
morpheans_morphesp-240            morpheans_morphesp-240           
unexpectedmaker_feathers2         unexpectedmaker_feathers2        
unexpectedmaker_feathers2_neo     unexpectedmaker_feathers2_neo    
unexpectedmaker_tinys2            unexpectedmaker_tinys2
```

Neradoc, 2021-09-08:
The way that section works is by reading this file referenced as `site.data.bootloaders` in the template.
https://github.com/adafruit/circuitpython-org/blob/master/_data/bootloaders.json

It contains the information on which port has a bootloader, and the conversion between the circuitpython board ID and the bootloader board ID (bootloader_id in the template). As far as I know it's entirely manually made (including the version number). But note that it currently contains 0 information on the ESP32-S2 port, so nothing would be shown for any board if you don't update that file too.

I think a basic "generic" version would be fine. Install: hold boot while the board boots, then use esptool and the combined.bin from the zip in the link. Update: reset then boot during purple status LED - within 1/2 second if no status LED. But it would be nice to have the ability to "overload" it in the board info with a bootloader section in `_board/the_board.md`.

durapensa, 2021-09-08:
Ahh, thank you. I edited that file for `lolin_s2_mini`, also adding `esp32s2` to `bootloaders`, for testing but overlooked the possibility of `bootloader_id` being different from the board's key in `boards`.

In that case, these boards should be an easy add:

```
adafruit_feather_esp32s2          adafruit_feather_esp32s2_nopsram 
adafruit_feather_esp32s2_tft      adafruit_feather_esp32s2_tftback_nopsram                         
adafruit_funhouse_esp32s2         adafruit_funhouse                
adafruit_magtag_29gray            adafruit_magtag_2.9_grayscale            
espressif_hmi_1                   espressif_hmi_devkit_1                
muselab_nanoesp32-s2_wroom        muselab_nanoesp32_s2_wroom       
muselab_nanoesp32-s2_wrover       muselab_nanoesp32_s2_wrover      
targett_mcb_wroom                 targett_module_clip_wroom        
targett_mcb_wrover                targett_module_clip_wrover    
```
So 22 boards in total.

PR #753 (2021-09-09): Add "Install/Repair UF2 Bootloader" section for ESP32-S2 boards
Fixes #752

Checked for typos, tested with `bundle exec jekyll serve`, and verified that download links are correctly generated on the Lolin S2 Mini page. Have not manually tested the other affected 21 pages for ESP32-S2 boards, but mappings in bootloaders.json should be correct.
closed 2021-09-09 by dhalbert