adafruit/circuitpython-org #755: improve board page detail and board searchability

durapensa, 2021-09-10:
I'd like to help improve the detail available on board pages, as well as board searchability. Nothing as crazy as DigiKey ;-) but I suspect some use of data/variables & includes could help, and I'm looking for guidance (new to Jekyll!).

For search, perhaps by using the same mechanism as "features" in `downloads.html`/`downloads.js` by adding a new data field, e.g. `data-mcu` and/or `data-components`, and including it in the nice `downloads-filter` section?

For includes for common components (including MCUs), `board.md` pages could use include tags directly (problematic for people making these pages?) or be conditionally included by the `download` layout based on variables (above) in `board.md` pages' front matter?

Could/should `bootloaders.json` (currently contains 149/234 boards) be generalized to accomplish any of this, rather than storing variables in `board.md` pages' front matter?

tannewt, 2021-09-10:
The js search method is definitely the way to go since the site is statically hosted.

I think you could store the data in either place, front matter or js. The js wouldn't need processing but I bet you could generate it from front matter using jekyll.

In the longer term I'd love to see a separate repo for board info like this similar to what I did with nvm.toml.

durapensa, 2021-09-11:
First attempt at #756, low-hanging fruit by using the existing MCU `family` data in `bootloaders.json`.

nvm.toml is nice!