adafruit/circuitpython-org #764: Gray download buttons may be confusing

dhalbert, 2021-09-19:
A user thought that an unstable release was not available for their board because the download buttons were gray. They thought they were grayed-out and did not even try to click on them. I wonder if we should pick a somewhat different color. The current color is #999999.

@jwcooper You may have a UX comment.

jwcooper, 2021-09-19:
Going to ask @tylerdcooper for this one.

Example:
[image]

maholli, 2021-09-21:
> A user thought that an unstable release was not available for their board because the download buttons were gray. They thought they were grayed-out and did not even try to click on them. I wonder if [...]

+1 we've encountered this with students as well!

jwcooper, 2021-09-21:
Let's just switch it all to the standard purple links. The ordering should be obvious enough that the top section is the latest release.

I can submit a PR tomorrow, unless someone beats me to it.

PR #767 (2021-09-22): Change .unstable and .unrecommended links to purple
- Fixes #764.

Change unstable and unrecommended download links to purple, to avoid confusion that the links are not functional.

I left the markers in, in case we want to change the link colors again, in the future. Also, I did not change the language dropdown away from gray.

[image]