adafruit/circuitpython-org #786: Make new unknown boards more accessible

dhalbert, 2021-11-15:
Currently, when an unknown board is added to CircuitPython, a page like this is generated:
[image]
All the links point to `.../unknown...`, etc., which is not too useful.

It would be nice if the board name (`trinket_m0` style, etc.) was used instead, so the new board's builds would at least be accessible before we got around to adding the new boards.

tannewt, 2021-11-15:
We can likely automate this with the data update for circuitpython.org when we do a release. It really should be something managed with a release.

makermelissa, 2021-11-15:
> We can likely automate this with the data update for circuitpython.org when we do a release. It really should be something managed with a release.

That's not a bad idea. It would probably look better than what we could do on the website alone.

dhalbert, 2022-04-26:
Fixed by #869, I believe.

closed 2022-04-26 by dhalbert

makermelissa, 2022-04-26:
It still points to `/unknown`, but that's the best we can do on this side with the limitations of liquid and jekyll needing to build the site pages ahead of time. I think the best solution would be to have stubs generated along with the `_data/files.json` as @tannewt suggested.