adafruit/circuitpython-org #809: add devkits nopsram variant

jepler, 2021-12-09:
Originally I thought maybe it would be the same firmware but it'll be two different ones. I ran out of time to update circuitpython-org before being "out of the office" for a week

makermelissa, 2022-02-15:
Does this board already exist in CircuitPython and if so, which one is it?

jepler, 2022-02-16:
```
espressif_esp32s3_devkitc_1_n8
espressif_esp32s3_devkitc_1_n8r2
espressif_esp32s3_devkitc_1_n8r8
```
these are the esp32-s3 devkit-c variants, should all be in the just-released 7.2.0-alpha.2.  The first is "nopsram", the second is 2MB PSRAM, and the third is 8MB PSRAM.  All have 8MB SPI flash.

makermelissa, 2022-02-16:
Ok, all of those are added in, so we should be able to close this.

closed 2022-02-16 by makermelissa