adafruit/circuitpython-org #816: [Security] Workflow adabot_cron.yml is using vulnerable action actions/checkout

Ale0x78, 2021-12-19:
The workflow adabot_cron.yml is referencing action actions/checkout using references v1. However this reference is missing the commit [a6747255bd19d7a757dbdda8c654a9f84db19839](https://github.com/actions/checkout/commits/a6747255bd19d7a757dbdda8c654a9f84db19839) which may contain fix to the some vulnerability. 
The vulnerability fix that is missing by actions version could be related to: 
(1) CVE fix 
(2) upgrade of vulnerable dependency 
(3) fix to secret leak and others. 
Please consider to update the reference to the action.

jwcooper, 2021-12-20:
Thank you for notifying us. We'll look into this further and upgrade as we can.

At this point, I'm not seeing this particular security issue directly impacting us. Are you able to reproduce this issue in our repository?

jwcooper, 2021-12-20:
After further review, there is no impact to this repository.

closed 2021-12-20 by jwcooper