adafruit/circuitpython-org #929: pimoroni badger 2040: shows 7.2.5 download, but is only in 7.3.0-beta.1

jepler, 2022-04-10:
Not sure why this occurred, it may be due to a problem with the circuitpython release process. see https://twitter.com/ukscone/status/1513253251403198475

dhalbert, 2022-04-10:
I had to make the releases by hand because of this failure, which keeps occurring:
for example: https://github.com/adafruit/circuitpython/runs/5861266045?check_suite_focus=true
Open the "Website" part; it looks like it succeeds, but it's really failing

I think I may have been in the wrong branch (`main` instead of `7.2.x`)  when I made the 7.2.5 release.

dhalbert, 2022-04-10:
Well, I redid it in the right branch, and it still didn't help.

dhalbert, 2022-04-11:
OK, I did it again, and this time it's OK. Not sure what the difference was. Badger now shows only 7.3.0-beta.1.

closed 2022-04-11 by dhalbert