adafruit/circuitpython-org #943: Bootloader section missing for some boards

caternuson, 2022-05-04:
For example, not showing up for Matrix Portal:

[image]

makermelissa, 2022-05-05:
It looks like we're only displaying bootloaders for the atmel samd boards and the nrf52840 chipsets. However, since this is an atmel-samd board, it should be displaying.

makermelissa, 2022-05-05:
It looks like it's missing the "bootloader_id:" line. I'll go through these and make sure they have these.

Neradoc, 2022-05-05:
Note, it's also missing on some ESP32S2 and ESP32S3 boards. For some of them it might be because there is no UF2 bootloader available, so that has to be checked, must some are definitely missing.

I ran a script:
```
adafruit_qtpy_esp32s2.md
ai_thinker_esp_12k_nodemcu.md
crumpspace_crumps2.md
electroniccats_bastwifi.md
espressif_esp32s2_devkitc_1_n4.md
espressif_esp32s2_devkitc_1_n4r2.md
espressif_esp32s3_box.md
espressif_esp32s3_devkitc_1_n8.md
espressif_esp32s3_devkitc_1_n8r2.md
espressif_esp32s3_devkitc_1_n8r8.md
espressif_esp32s3_devkitm_1_n8.md
espressif_esp32s3_usb_otg_n8.md
espressif_kaluga_1.3.md
franzininho_wifi_wroom.md
franzininho_wifi_wrover.md
gravitech_cucumber_m.md
gravitech_cucumber_ms.md
gravitech_cucumber_r.md
gravitech_cucumber_rs.md
hexky_s2.md
hiibot_iots2.md
lilygo_ttgo_t8_s2_st7789.md
lilygo_ttgo_t8_s2.md
odt_pixelwing_esp32_s2.md
unexpectedmaker_feathers2_prerelease.md
```

And it's present in some ESP32C3 boards that shouldn't have it, since they can't use the UF2 bootloader.

```
adafruit_qtpy_esp32c3.md
microdev_micro_c3.md
```

makermelissa, 2022-05-05:
Fair enough. I'll reopen.

PR #944 (2022-05-05): Add missing atmel-samd and nrf52840 bootloader IDs
Fixes #943.
closed 2022-05-10 by makermelissa