adafruit/circuitpython-org #1001: ESP32-S3: Add information on using the UF2 bootloader

kmatch98, 2022-07-23:
The ESP32-S3 board descriptions explain how to install the UF2 bootloader.

I recommend adding details on how to trigger into UF2 bootloader mode:

1. Click RESET
2. While purple LED is lit, click BOOT. (LED will flash red and then remain green.)

dglaude, 2022-07-24:
Thank you for this.
I was desperated and every time I have to use esptool to upload the bin file... just because I did not knew that trick.
Thanks.
I guess that works for most Espresif... it deserve to be known.

dhalbert, 2022-07-24:
@kattni How about we add specific info on the "CircuitPython Quickstart" and "Installing UF2 Bootloader" Learn Guide templates that double-clicking on the Espressif boards involves waiting for the purple. I myself didn't know this a few months ago and was stymied until I was told.

kattni, 2022-07-26:
I updates all the guide templates and the applicable guides to reflect the purple LED bootloader indicator.

PR #1005 (2022-07-26): Improve ESP32-nn bootloader installation instructions
Fixes #1001.

Caution users to refer to board documentation to update or install UF2 bootloader.
Describe how to enter UF2 bootloader.
closed 2022-07-26 by makermelissa