adafruit/circuitpython-org #1012: Board Checker Script Enhancement Idea

makermelissa, 2022-08-05:
An easy way to avoid stuff like #1010 would be to have it check that the board_id matches the filename. This will probably need some valid exceptions added like unknown though.

FoamyGuy, 2024-10-28:
I started this on a branch here: https://github.com/FoamyGuy/circuitpython-org/tree/check_boardid_filename

One thing that we'll need to decide is if / how to deal with the ones that are already merged in and would not pass this check, here is the current list: 

```
# core: 
board_id filename mismatch: thingpulse_pendrive_s3 != ThingPulse_PendriveS3
board_id filename mismatch: adafruit_feather_rp2040_rfm != adafruit_feather_rp2040_rfm69
board_id filename mismatch: challenger_rp2350_bconnect != challenger_rp350_bconnect
board_id filename mismatch: challenger_rp2350_wifi6_ble5 != challenger_rp350_wifi6_ble5
board_id filename mismatch: circuitart_zero_s3 != circuitart_esp32s3_zero
board_id filename mismatch: crcibernetica-ideaboard != crcibernetica_ideaboard
board_id filename mismatch: flipperzero_wifi_dev != flipper-zero_wifi_devboard
board_id filename mismatch: heiafr_picomo_v2 != picomo_v2
board_id filename mismatch: seeed_xiao_esp32_s3_sense != seeed_xiao_esp32s3_sense
board_id filename mismatch: devkit_xg24_brd2601b != silabs_devkit_xg24_brd2601b
board_id filename mismatch: explorerkit_xg24_brd2703a != silabs_explorerkit_xg24_brd2703a
board_id filename mismatch: sparkfun_thingplus_matter_mgm240p_brd2704a != sparkfun_thing_plus_matter_mgm240p
board_id filename mismatch: unexpectedmaker_feathers3 != unexpectedmaker_feather3

# blinka:
board_id filename mismatch: adafruit_feather_rp2040_rfm != adafruit_feather_rp2040_rfm69
board_id filename mismatch: nanopi_neo != nanopi_neo_h3
board_id filename mismatch: raspberry_pi_3bplus != raspberry_pi_3b

```

makermelissa, 2025-12-15:
This is already done. It is part of **tools/check-boards.py** here: https://github.com/adafruit/circuitpython-org/blob/main/tools/check-boards.py#L17-L29.

securelyfitz, 2026-09-17:
howdy. just ran into this on a0e19fa3a6a35d8bd6395b41d2e9c987eedf2b40 and check_boards does *not* check to make sure board id and filename match.

makermelissa, 2026-09-17:
> howdy. just ran into this on [a0e19fa](https://github.com/adafruit/circuitpython-org/commit/a0e19fa3a6a35d8bd6395b41d2e9c987eedf2b40) and check_boards does _not_ check to make sure board id and filename match.

Yeah, that would be a good thing to check.

securelyfitz, 2026-09-17:
ideally... it'd be cool to check against the board_id and filename against a valid "id" in _data/files.json too...

makermelissa, 2026-09-17:
> ideally... it'd be cool to check against the board_id and filename against a valid "id" in _data/files.json too...

I think the only issue with that is that we do allow boards to be submitted prior to them being merged into CircuitPython. If the board is not in the data file, the download links just don't appear until it is added.

securelyfitz, 2026-09-17:
Makes sense. 
In the end my issue was that i was not consistent with the board_id i used in the circuitpython repo, not that the board_id and filename didn't match in the circuitpython-org repo.

makermelissa, 2026-09-17:
I added the additional checks and fixed a few other boards in #1812. I'm not sure why I closed it before, it wasn't actually finished.

securelyfitz, 2026-09-17:
That was quick. Thanks!