adafruit/circuitpython-org #1015: Consider adding a badge for product availability next to adafruit product links

jepler, 2022-08-09:
I was thinking about circuitpython.org and status badges and wondered whether it would be good to add badges next to product links that show availability. Here's a mock-up:
[image]

[shields.io has a sort of API](https://shields.io/endpoint) to assist in this, though it requires that you supply an endpoint that returns json in an expected format. Using shields.io gets nice stuff by default, like the "images" (or svg files) are served from their cdn with a default cache lifetime of 300 seconds. In order to use this "endpoint style" the store's product API would need to be changed to provide data in the right format.

Alternatively, the circuitpython.org page could first make an API request to the store and then insert the right badge, still possibly using shields.io with a constructed URL like https://img.shields.io/badge/Availability-In%20Stock-green

One reason to consider NOT doing this is that anything other than "in stock" would stop at least some folks from clicking through, reducing "notify me" sign ups, related item purchases, etc.

Please consider this just an idea, no hurt feelings on my part if it's not worth adopting. Feel free to close this up if so.

jwcooper, 2022-08-09:
I think we don't want to go this route due to having to support every board and manufacturer as circuitpython.org isn't Adafruit specific.

The maintenance load of supporting many product in-stock APIs would be too cumbersome for not enough value.

In addition to that, we'd have to update the board metadata of every board to either provide more information (in stock api endpoints), or do some precarious parsing of our generated HTML to poll endpoints that would be specific for every board.

Most of Adafruit products are usually in stock, and if they are not, many users would want to just get notified when they get in stock anyways.

tannewt, 2022-08-10:
I don't think we need to support every manufacturer on cp.org. It is open source so interested non-Adafruit folks can diy.

We wouldn't need extra code here if we make a shields compatible endpoint from the adafruit shop. Here it'd just be a SVG in the body.

We could do Adafruit specifically but it doesn't seem like we want to. IMO saying things are in stock is motivating even if they are typically in stock. (Because folks expect things to be out of stock.)

closed 2022-08-15 by tannewt