adafruit/circuitpython-org #1016: Hard to find ESP32 board (like huzzah32)

dglaude, 2022-08-12:
Right now it seems difficult to find and know that CircuitPython can run on plain old ESP32.

There are no board present in the list:
* Feather ESP32 Huzzah v2 https://www.adafruit.com/product/5400 where apparently you can find on the S3: https://adafruit-circuit-python.s3.amazonaws.com/index.html?prefix=bin/adafruit_feather_huzzah32/
* Odroid Go that is visible here: https://adafruit-circuit-python.s3.amazonaws.com/index.html?prefix=bin/hardkernel_odroid_go/
* Legacy "Adafruit HUZZAH32 – ESP32 Feather Board" https://www.adafruit.com/product/3405 and the binary is no where to be found, or maybe in the artefact from some PR 
(There might be other board I don't know about, but how would I know what I don't know)

In the download section there is not ESP32 MCU to choose in the filter criteria.
(There might be other place that need tweeking for a new MCU to be visible in circuitpython.org)

So if someone not following very closely want to know it is now possible to run on ESP32 and want to try on some well known board... it is made very hard for that person.

I am pretty sure there are a lot of conference badge or other board people already have that could benefit from ESP32 support and act as promotion material for CircuitPython. With new maker/hacker event taking place left and right, this is the right time to advertise and make things more easier.

ladyada, 2022-08-12:
this is on purpose right now - ESP32 is still just started working and many bugs are being fixed and workflow is being documented :)

dglaude, 2022-08-12:
This is really odd, there are other MCU/boards that are very beta, far from totally supported, and have their page present since day 1.

If you want people not to be deceived and have a bad user experience while using alpha/beta version or early support, you could work reversely on CircuitPython.org. Rather than to display everything by default, only show the 250 or 300 solid build, but have a filter that say "Show every board including very alpha".
By the number you will be able to claim as many board or more, but direct to the best user experience...

closed 2022-08-12 by ladyada