adafruit/circuitpython-org #1031: Enhancement of the filter for not a having a feature.

dglaude, 2022-08-23:
Right now you can filter for board with feature X, but you cannot filter for board that do not have feature X.
I used the "display" filter to identify board that might pretend to have a display, but not really.
I would like to use the "no display" filter to identify (visually) board with a display but that forgot to advertise it.

There could be other use for such negative filtering, and I have no clue how to make a user interface that make that easy.
Right now my use case is of limited interest and might make things too complex for few benefits.

makermelissa, 2022-08-23:
Hmm, this could be interesting. We'd have to think about how the UI would function.

FoamyGuy, 2024-10-28:
This feels to me like a fairly niche / maintenance focused feature. 

My initial inclination would be to implement it without a visual UI.  The current filter checkboxes add the values checked to to the URL as parameters e.g. `?features=Display` which then makes it behave like a search term limiting the list to only device which have the specified features. 

I would propose to modify the JS that handles the limiting based on features functionality to support a negative modifier in the value so a value of `-Display` would denote "not Display".

In order to use this feature the user would have to know of it's existence and modify the URL by adding the minus signs to whichever features they're interested in.

FoamyGuy, 2024-10-28:
I poked at this for a little bit but did not end up finding the right way or place to change the filtering behavior to invert it for the "not" effect. I do think it's possible, but might be more complex than my original comment makes it sound even if we do skip the visual UI.