adafruit/circuitpython-org #1032: is a HUB75 matrix connect a display? => New feature?

dglaude, 2022-08-23:
There are at least two boards with a HUB75 matrix LED interface:
* matrixportal_m4.md
* pimoroni_interstate75.md

The Matrix Portal from Adafruit advertise to have a "Display" as feature.
Where the Interstate 75 from Pimoroni does not say it has that feature.

So I don't know witch one to modify.

Maybe this is a new feature and that rather than to say "Display" there should be something called "HUB75" or a more friendly name. Then both board could advertise that feature.

dhalbert, 2022-08-23:
We might say "display interface" or "RGB matrix interface". Not sure if the latter is too specific. @makermelissa do you have an opinion here?

makermelissa, 2022-08-29:
For the terms, let's go with:
Built-in Display
External Display Connector

"External Display Connector "could be added to the bare metal Raspberry Pi SBCs and similar boards as well.

dglaude, 2022-09-19:
I made a vote in Discord Chat during the weekly meeting and the final result were "External Display" as a winner:
1 :adabot: Display interface
4 :adafruit: External Display
1 :cappy: External Display Connector
0 :gus: Intended for External Display
0 :mho: Optional Display
0 :blinka: Display Friendly
2 :toebeans: Display Ready

makermelissa, 2022-09-20:
Sounds fair to me. :) Unfortunately I was out sick yesterday so wasn't able to vote, but since it wouldn't have made a difference, I'm ok with going with "External Display".

dhalbert, 2022-09-20:
How about "Display Capable" ?

dglaude, 2022-09-26:
The name has been selected (it is still possible to change it) but more importantly the board that need it have been tagged with PR such as #1030 and #1029 .

closed 2022-09-26 by dglaude