adafruit/circuitpython-org #1035: Feature request: Show board features on board pages

dglaude, 2022-08-27:
Maybe on a board page like https://circuitpython.org/board/circuitplayground_express/ we could display the features of that board in "little boxes".

So for CPE it will display those two items from the metadata:
  - Speaker
  - Solder-Free Alligator Clip

Graphically it could be a bit like the display of selected filter: https://circuitpython.org/downloads?features=Solder-Free+Alligator+Clip&features=Speaker 

It can act as a summary for normal user of the site, but my use case is more to verify error (or false claim) on the feature of a board.
By inspecting the picture and the text it would be easy to spot problem with a board metadata.

I am not sure if it should just be a visual indication, or maybe it could be a button that trigger something related to the search, like making a search on that criteria. The use case would be that you are on a board that has feature XXX and you click on XXX to find all the board with that feature https://circuitpython.org/downloads?features=XXX

makermelissa, 2022-08-27:
Yeah, that makes a lot of sense. :)

DJDevon3, 2022-11-05:
I 2nd this motion.

PR #1513 (2024-10-28): show features on board download page
Resolves: #1035 

I added them below the built-in modules inside of purple "tag" elements. 

[image]

The do not currently do anything when clicked. 

This was my interpretation of the "little boxes" mentioned in the linked issue. I am open to feedback on the style or placement of them.
closed 2024-10-28 by dhalbert