adafruit/circuitpython-org #1060: Hacktoberfest appears twice in Issues sort dropdown

prcutler, 2022-10-04:
On the Contribution page:  https://circuitpython.org/contributing/open-issues?label=hacktoberfest

The issue sort drop down shows Hacktoberfest twice.  Kattni wasn't sure if it was related to the label or an org issue.

[image]

prcutler, 2022-10-04:
It might be a capitalization issue as I see two different tags in the list, with and without a capital H
[image]

makermelissa, 2022-10-06:
Good catch. I wonder if just adding the downcase filter to https://github.com/adafruit/circuitpython-org/blob/main/contributing/open_issues.html#L16 might fix it.

PR #1062 (2022-10-06): Fix duplicate Hacktoberfest label
Fixes #1060.
closed 2022-10-06 by makermelissa