adafruit/circuitpython-org #1091: Adabot submodule needs to be fixed and repinned

tekktrik, 2022-11-07:
The cron the updates the contributing information is failing because Adabot's current method of using validators is failing for Python 3.11.  I've submitted https://github.com/adafruit/adabot/pull/309 for this - hopefully that's the only change needed.  After that's merged, the `adabot` submodule should be updated here.

tekktrik, 2022-11-07:
Looks like this wasn't enough - I think it may be worth pinning the Python version similar to the way Adabot does to Python 3.9.  If so, should it be penultimate (3.10) or the same as Adabot (3.9)?

makermelissa, 2022-11-07:
Sure. In here? https://github.com/adafruit/circuitpython-org/blob/main/.github/workflows/adabot_cron.yml

tekktrik, 2022-11-07:
Yup! For reference, the BigQuery package adabot now needs isn't avialable for Python 3.11 yet.

tekktrik, 2022-11-07:
Linking https://github.com/googleapis/python-bigquery/issues/1386 so we can track it potentially

makermelissa, 2022-11-09:
I see there were a couple of related PRs. Is this fixed now?

tekktrik, 2022-11-09:
Yup!  Thanks for catching!

PR #1092 (2022-11-07): Update to adabot with PyGithub 1.57
Resolves #1091
closed 2022-11-09 by tekktrik