adafruit/circuitpython-org #1098: Make /board alias or redirect to /downloads ?

dglaude, 2022-11-13:
Should https://circuitpython.org/board be "an alias of" / "a redirect to" https://circuitpython.org/downloads ?

The idea is that when you are on a board and you want to check other board, you can shortcut the URI to do that

https://circuitpython.org/board/raspberry_pi_pico_w/
=>
https://circuitpython.org/board/
or
https://circuitpython.org/board

(maybe I am the only one doing/trying that...)

jwcooper, 2022-11-14:
If we do this, I'd prefer to also redirect /board/board-name to /boards/board-name. The redirect to all the boards would then go from /downloads to /boards. This would make more sense if we ever created a circuitpython.org json REST api as well. https://circuitpython.org/boards/raspberry_pi_pico_w.json, etc.

In hindsight, this what the endpoints probably should have been when the site was first built, but we went with a more traditional /downloads endpoint that kind of stuck.

If we do this though, it brings along maintenance considerations as we will always have to maintain these redirects. In the case of /board -> /boards, as well as /board -> /downloads. 

I'm not entirely convinced it's worth that cost yet (I haven't looked into how it's done in jekyll/github pages either).

dhalbert, 2022-11-14:
Jekyll has a plugin: https://github.com/jekyll/jekyll-redirect-from

https://stackoverflow.com/questions/9276817/301-redirect-for-site-hosted-at-github, one of many discussions on this