adafruit/circuitpython-org #1102: Website not updating

tekktrik, 2022-11-14:
The CI seems to be running and passing but the website isn't updating and still lists the 11/11/22 as the last update.

dhalbert, 2022-11-14:
Maybe related to this?

https://github.com/adafruit/circuitpython-org/actions/runs/3457109483

tekktrik, 2022-11-14:
Maybe, it looks like it ran last night.  If no one manually retriggered it it looks like its resolved!

FoamyGuy, 2022-11-14:
I noticed this stacktrace in one of the actions task for the scheduled jobs:
https://github.com/adafruit/circuitpython-org/actions/runs/3460106060/jobs/5776269092

it's inside of `adabot.update_cp_org_libraries step
```
Running circuitpython.org/libraries updater...
Run Date: 14 November 2022, 09:24AM
 - Report output will be saved to: /home/runner/work/circuitpython-org/circuitpython-org/bin/adabot/libraries.v2.json
Unhandled exception list index out of range
Traceback (most recent call last):
  File "/home/runner/work/circuitpython-org/circuitpython-org/adabot/adabot/update_cp_org_libraries.py", line 234, in main
    errors = validator.run_repo_validation(repo)
  File "/home/runner/work/circuitpython-org/circuitpython-org/adabot/adabot/lib/circuitpython_library_validators.py", line 285, in run_repo_validation
    errors.extend(validator(self, repo))
  File "/home/runner/work/circuitpython-org/circuitpython-org/adabot/adabot/lib/circuitpython_library_validators.py", line 1266, in validate_actions_state
    if workflow_runs[0].conclusion != "success":
  File "/opt/hostedtoolcache/Python/3.9.15/x64/lib/python3.9/site-packages/github/PaginatedList.py", line 49, in __getitem__
    return self.__elements[index]
IndexError: list index out of range

```

I'm not sure if this is related, but I think the scheduled actions tasks are the ones updating the contributing page rather than the triggered on PR / push ones.

tekktrik, 2022-11-14:
Yeah that should be the step that gathers all the information needed for generating that page.  If it fails, I assume it uses the existing (previous day's) information when regenerating the website since that file never gets overwritten.

That's also my code from adabot that's causing it to fail so I can wrap it in a try/except - looks like it trips up checking the CI status if the CI has never run, so I can patch that.

tekktrik, 2022-11-14:
Alright, submitted the fix to adabot; I'll also update it as a submodule here once that's approved and merged.

PR #1103 (2022-11-14): Update adabot to fix error when building contributions
Fixes #1102
closed 2022-11-14 by tekktrik